ozzmaker/BerryIMU #6: Correct Y readings with X==90 degrees?

kramer65, 2017-03-27:
When I run the code from [python-BerryIMU-gryo-accel-compass/berryIMU.py](https://github.com/mwilliams03/BerryIMU/blob/master/python-BerryIMU-gryo-accel-compass/berryIMU.py) and tilt the BerryIMU 90 degrees to the right I would expect Y to be around 0, but instead it goes to about 100. 

I asked [a question about this behaviour on robotics.stackexchange](http://robotics.stackexchange.com/questions/11870/why-does-one-imu-axis-influence-another), and [the answer I got](http://robotics.stackexchange.com/a/11881/1050) was that the problem was "_estimating angles in regions of instability_" in which they suggest to "_modify unstable equations_".

Can you confirm that this is the problem I'm having? Furthermore, I'm kinda lost in the maths they present, but do you have a suggestion how to incorporate this in your code?

mwilliams03, 2017-08-08:
This is the formula you can use to get around this issue
  pitch = atan2(x, sqrt(y * y) + (z * z));
  roll = atan2(y, sqrt(x * x) + (z * z));
  pitch *= 180.0 / PI;
  roll *= 180.0 / PI;

closed 2017-08-08 by mwilliams03

russellhoff, 2017-12-22:
Which are the units for roll and pitch values? I guess they're in degrees...