ozzmaker/BerryIMU #7: Compile Error

Ssabaker, 2017-07-01:
I am currently trying to work my way through the tutorials, however, I can't get the three tutorials to compile due to various errors.  The first and the third have errors where I2C_SMBUS_BLOCK_MAX is undeclared while the second one can't find the directory for "SDL.h". Is there anything that I might be missing here? Thank you

mwilliams03, 2017-07-03:
Which program are you using where you get these errors?

Ssabaker, 2017-07-03:
I'm trying to run it through the terminal by following the read me's. What program should i be using?

mwilliams03, 2017-07-04:
If you are just after angles and compass heading, then use this code;
https://github.com/mwilliams03/BerryIMU/tree/master/python-BerryIMU-gryo-accel-compass

Ssabaker, 2017-07-04:
Now I'm getting an error for line 36: bus = smbus.SMBus(1)
But it's importing smbus

Ssabaker, 2017-07-04:
went back and reinstall the smbus and its working now

Ssabaker, 2017-07-04:
appreciate the help, thank you

closed 2017-07-05 by mwilliams03