ozzmaker/BerryIMU #17: used this with MPU9250, modified accordingly and works but issue with heading

bbelmar, 2020-04-16:
Hello there, first of all thank you so much for this amazing work! you are a life saver.

I used this and modified to get the readings from MPU9250. Accel works as expected, I was expecting the gyro to read similar magnitudes as the accel but the magnitudes are way smaller, even if I tilt the sensor -90Deg in the Y axis , the gyro measures only approx -6., but the accel measures approx -90. I thought the code was turning gyro information in Degrees? The accel in X goes crazy but i dont know if that is an expected reading.

Accel in X=  146.01047352241366 
 Accel in Y=  -94.48802755114491   ##this is when i rotated the sensor to the left 90 deg approx
 GyroX =  0.0958970146179199 
 GyroY =  -6.615839959945679  ## shouldn't this be same reading as accel in Y?
 GyroZ =  0.010706434631347735 
 Heading =  0.4136347527893839 
 Compensated Heading =  -496.89575038831293

And lastly, as you can see the heading is not responding well... 
If i rotate the sensor 180 deg, it goes from approx 25-> 0 -> 40 as i rotate...

I know im asking a lot, but if you have the time to reply would be amazing. Thank you so much!

bkfichter, 2021-03-17:
I don't get the "compass" either. The heading values don't match anything, at least using the demo code. Did you get anything working?
And I see by the lack of response for your question, I may have made a mistake in purchasing this unit as there doesn't seem to be support.

bkfichter, 2021-03-17:
Did you ever get this working? For the BerryGPS-IMU v4, I have been using the python versions of the calibration and "compass" and the headings are nothing usable. And tilt compensation doesn't do anything usable either.

peterp-ozzmaker, 2021-03-17:
> Did you ever get this working? For the BerryGPS-IMU v4, I have been using the python versions of the calibration and "compass" and the headings are nothing usable. And tilt compensation doesn't do anything usable either.

Have you tried to graph out your results after calibration to confirm that the calibration is correct?
[Calibration guide](https://ozzmaker.com/compass3/)

bkfichter, 2021-03-17:
Well, I did run calibrateBerryIMU.py (multiple times) and took the results and applied to berryIMU.py. I just checked to make sure the code was implementing the calibration results. It is. I haven't actually graphed the results. I will do this today. I'm not sure what this will give me if I am already running the calibration script and applying the results.