ozzmaker/BerryIMU #21: BerryGPS-IMU v4 "python-BerryIMU-gyro-accel-compass"

bkfichter, 2021-03-17:
I am new at this so after reading and going through the tutorials and trying out the sample python code (/python-BerryIMU-gyro-accel-compass/berryIMU.py), I'm not seeing this as a usable compass. The values are off, even with calibration. And the calibration values change each time you run the calibration script anyway. The heading values even change for the same orientation if you continue to rotate the device. I can't even tell you which direction is "forward" on the unit because the values don't match any compass direction. And there is no tilt compensation even though there seems to be tilt compensation correction in the code.

So my question is, has anyone ever gotten this thing to behave like a real functioning compass like you have on a smart phone or is this unit not capable of doing it? Again, I am not an expert in this area.

mwilliams03, 2021-03-18:
Have you tried to graph out your calibration results to confirm the calibration is correct?
Have a look at this [guide](https://ozzmaker.com/compass3/)

bkfichter, 2021-03-21:
I have. I am now starting from scratch and going through the tutorial from the start using the C code. When using compass_tutorial01, I am getting values from roughly 90-180. It occurs to me I don't now what UP or FORWARD is on the BerryGPS-IMU v4. I have it plugged into the headers on a Pi 4. The battery and blox chip are on top. I see the X,Y, and Z indicator, but I'm not sure how to read that. In this configuration, what is UP and what is FORWARD? Is the indicated Y forward? Still doesn't explain the 90-180 (repeated 2x when rotating) but let me get the orientation correct first.

mwilliams03, 2021-03-22:
that is the correct direction.
try working with [this ](https://github.com/ozzmaker/BerryIMU/blob/master/python-BerryIMU-gyro-accel-compass/berryIMU-simple.py )code as it include calibration and tilt compensation

bkfichter, 2021-03-22:
I think this is the exact same code that I started this thread with.  It was producing wrong values even after calibration. That's why I started over with the 1st tutorial,  to see if I missed something somewhere.

bkfichter, 2021-03-24:
I have spent HOURS on this. I have done the python. I have done the C. I have calibrated to a perfect circle dead center. I have applied those max/mins to the BerryIMU.py. I have added the declination. I have done it all. The best I have gotten is comparing to my phone compass (which has been compared to a physical compass and is near accurate) is:
Compass          BerryGPS-IMU
0                       250
90                     358
180                   80
270                   165

So it is a rough 90 off. Which is not the forward as validated above. This can be adjusted in code and fine for a rough prototype, but not for anything real. I'm guessing at this point this is good as this unit gets using this code and for more accuracy and stability, I need either better calibration process and code and/or a better chip (?).

mwilliams03, 2021-03-24:
Can show me the results of the below test;
Use an original version of this;
[https://github.com/ozzmaker/BerryIMU/blob/master/python-BerryIMU-gyro-accel-compass/berryIMU-simple.py](https://github.com/ozzmaker/BerryIMU/blob/master/python-BerryIMU-gyro-accel-compass/berryIMU-simple.py)

Add your calibration values to the script

Right after these lines;
```
    MAGy = IMU.readMAGy()
    MAGz = IMU.readMAGz()
```
add;
`print ("ACC %i %i %i GYR %i %i %i MAG %i %i %i  " % (ACCx,ACCy,ACCz,GYRx,GYRy,GYRz,MAGx,MAGy,MAGz))`
Disable all other print statements.

Start the script.
When it starts to output data, rotate the IMU(or Pi) 360 degrees and then stop the script.

please respond with the data and your calibration values.

bkfichter, 2021-03-25:
Thanks @mwilliams03 

From your instructions, I assumed rotate on the Z axis only. However I provided a file with rotation all axis also.

magXmax = 933
magYmax = 2570
magZmax = 1567
magXmin = -2348
magYmin = -784
magZmin = -1604

[output_AllRotation.txt](https://github.com/ozzmaker/BerryIMU/files/6201944/output_AllRotation.txt)

[output_ZRotation.txt](https://github.com/ozzmaker/BerryIMU/files/6201937/output_ZRotation.txt)

mwilliams03, 2021-03-29:
Your output looks strange.
This is the output from a test i just did.  Vanilla code and just added calibration values.
Below is the compass heading graphed out. One turn on the Z axis and it goes between 0 and 360
[image]
Here is your output (including calibration)
[image]

Can you confirm that the changes you have made to the code are only  to add the calibration values and the print statement to show the extra output.
Did you also only do one 360 degree turn in your test?

Can you do another test with the latest code from git, add the calibration values and print statement and then do another 360 rotation on the Z axis.
Can you also show the output of few hundred lines of the IMU not moving

bkfichter, 2021-03-30:
Did you use "output_ZRotation.txt"?

mwilliams03, 2021-04-06:
Yes, i used the file with Z rotation

bkfichter, 2021-04-08:
@mwilliams03 
I redid everything and made sure the GPS/IMU was perfectly level with one perfect rotation on the Z axis.
Here is still, z_Rotate and the python file I used (renamed .txt in order to upload here.)

[still.txt](https://github.com/ozzmaker/BerryIMU/files/6280777/still.txt)
[z_Rotate.txt](https://github.com/ozzmaker/BerryIMU/files/6280778/z_Rotate.txt)
[berryIMU-simple.txt](https://github.com/ozzmaker/BerryIMU/files/6280779/berryIMU-simple.txt)

bkfichter, 2021-04-15:
@mwilliams03

bkfichter, 2021-04-15:
@mwilliams03 
Any new thoughts? Thanks

mwilliams03, 2021-04-19:
Sorry, i didnt noticed you responded.
I pumped your z_rotate.txt output into Excel  which has all the formulas to calculate the headings.. and it looks correct to me.
It starts at around 75, goes 360 and circles  back to 75
[image]

I have uploaded the xls for you to review.(https://github.com/ozzmaker/BerryIMU/files/6335555/tilt.compensation.formula.with.calibration.xlsx)

bkfichter, 2021-04-19:
@mwilliams03 
Thanks. How do I modify the excel to get the graph. I want to graph the "north.txt" file. That file starts with the unit facing north. I want to see if the calculated heading matches north. This is where my issue started. When the unit is facing north, the values were about 90 degrees off.

bkfichter, 2021-04-19:
@mwilliams03 
So i just copied and pasted the stuff and it worked. The value for NORTH is around 265 so the same general error.
Is it common to have to adjust in code for the misaligned values coming off the unit (I am using https://github.com/ozzmaker/BerryIMU/issues/21#issuecomment-803642436 as north)? Could the unit be mounted incorrectly? Is there any other reason it is off?

bkfichter, 2021-04-21:
@mwilliams03

bkfichter, 2021-04-23:
@mwilliams03 
 Hey sorry to be a bother, but so close to determining if this is going to be a valid solution.  Any thoughts on the previous post?

mwilliams03, 2021-04-29:
okay.. i misunderstood your question.   North is on the right hand side of the PCB, so what you are seeing is correct.