ozzmaker/BerryIMU #25: Type Conversion error in base code for ArduinoBerryIMUv3

RubYourDuck, 2021-12-08:
Copy pasted folder from https://github.com/ozzmaker/BerryIMU/tree/master/arduino-BerryIMU and ran it and faced this issue. No changes were made in the code. Error below:

C:\Users\F\arduino-BerryIMU\IMU.cpp: In function 'void enableIMU()':
C:\Users\F\arduino-BerryIMU\IMU.cpp:52:59: warning: invalid conversion from 'byte {aka unsigned char}' to 'byte* {aka unsigned char*}' [-fpermissive]
         readFrom(LSM6DSL_ADDRESS,LIS3MDL_CTRL_REG3,1, temp);
                                                           ^
C:\Users\F\arduino-BerryIMU\IMU.cpp:20:6: note:   initializing argument 4 of 'void readFrom(int, byte, int, byte*)'
 void readFrom(int device, byte address, int num, byte buff[]) {

rtockes14, 2024-12-02:
Any update on this ^^   I'm This is a few years old now and the issue still persists.

mwilliams03, 2024-12-03:
this should now be fixed

closed 2025-04-08 by mwilliams03