ozzmaker/BerryIMU #28: Help with BerryGPS-IMU

ABUCKY0, 2025-02-19:
I'm unable to sign up for the Ozzmaker forums (presumably because GMail is bouncing your emails but I digress), so I'm asking here instead.

A few weeks ago my BerryIMU stopped working and shows -- for every i2c (bus?) listed in `i2cdetect`. 

Here's a video of what I mean: I stopped it mid way through because it doesn't need a 5 minute video to explain that every thing is --. Even with other i2c devices connected it shows `--`. I've tried only plugging the BerryGPS-IMU into the Pi to no avail. 

https://github.com/user-attachments/assets/144dc547-5af7-4f36-bbaf-908c0b096571

Here's how I have it wired 

[image]

mwilliams03, 2025-02-21:
this usually indicates a connection issue on the i2c bus. I doublechecked your  QWIIC wiring and it looks good.
Do you have anything else using i2c?
have you made any modifications to the PCB of the BerryGPS-IMU?

ABUCKY0, 2025-02-21:
I do have another device connected to i2c but I have tested without it connected and still had the same problem. I have not modified the BerryIMU

mwilliams03, 2025-02-21:
And if you have it sitting on top of the GPIO pins, you get the same issue?
Has i2c  ever worked for the BerryGPS-IMU?

ABUCKY0, 2025-02-21:
> And if you have it sitting on top of the GPIO pins, you get the same issue? 
Wdym sitting on top of the GPIO pins? 
> Has i2c ever worked for the BerryGPS-IMU?
Yeah

mwilliams03, 2025-02-22:
connected like this, rather than via a QWIIC cable

[image]

ABUCKY0, 2025-02-22:
I don't have the header thing that solders to those 10 pins on the imu

mwilliams03, 2025-02-22:
last question, does the other i2c device function correctly when the BerryGPS-IMU is not connected?

ABUCKY0, 2025-02-22:
It does not

mwilliams03, 2025-02-22:
this indicates a problem with the Pi.  Do you have another Pi to test with?

mwilliams03, 2025-02-22:
okay... then, can you send proof of purchase to sales@ozzmaker.com and we can organise for it to be replaced

closed 2025-02-28 by ABUCKY0