ozzmaker/BerryIMU #29: In arduino-BerryIMU, the heading, and other values, are grossly inaccurate.

bulldog10000, 2025-04-08:
Tested out this code with **BerryGPS-IMU v4** connected to an Arduino MKR WAN 1310 via I2C.

The values showed up, but were clearly wrong. Most noticeably, the _heading_ field was always between 0 and 90 degrees, no matter which way I turned it. The gyroscope readings were just monotonically increasing integers. And on closer inspection of the raw data values, they were _always positive_. But accelerometer and magnetometer readings should definitely be signed values. Still learning about gyroscopes readings, but assuming it is the same for those readings.

Easy fix. All that needs to be done to fix it is to cast the composite high/low bytes value to a 'short', instead of an 'int'.

I created a [pull-request](https://github.com/bulldog10000/BerryIMU/tree/patch-1).

This is my first ever attempt at contributing to GitHub, so please let me know if I'm doing it wrong.

mwilliams03, 2025-04-08:
not there yet, try to follow this
[https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork](url)

bulldog10000, 2025-04-08:
> not there yet, try to follow this [https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork](url)

Thanks.

[How about this?](https://github.com/ozzmaker/BerryIMU/pull/30)

mwilliams03, 2025-04-08:
merged !
Thanks